Xinyuan-LilyGO/LilyGo-Display-IDF #5: Referenced version of ESP-IDF not available

jorgie0, 2024-03-16:
Hi
The referenced version of ESP-IDF is not available for download.

`The example relies on esp-idf V5.3.0, Please see the [Getting Started Guide](https://docs.espressif.com/projects/esp-idf/en/latest/get-started/index.html) for full steps to configure and use esp-idf to build projects.`

lewisxhe, 2024-03-16:
Look here, the link has changed
https://docs.espressif.com/projects/esp-idf/en/latest/esp32s3/get-started/index.html

jorgie0, 2024-03-16:
@lewisxhe  Then I am missing something I cannot see V5.3 of ESP IDF.
I am looking here:
https://github.com/espressif/esp-idf/tree/master

[image]

lewisxhe, 2024-03-16:
[image]

jorgie0, 2024-03-16:
@lewisxhe my point if that version 5.3 cannot be found where the red arrow is pointed. The paragraph I quoted stated that it relies on esp-idf V5.3.0,
So please help me understand what I am doing wrong?

krystalmonolith, 2024-03-19:
@jorgie0 You are correct ... As of now (March 19 2024) I see no "5.3.x" tag or release of the SDK... The last one I see is 5.2.1 on their tags page at https://github.com/espressif/esp-idf/tags . There have been many commits since then and I assume @lewisxhe intends that you should use them by working from the ```master``` branch, which will  eventually become 5.3.x .

lewisxhe, 2024-03-20:
Yes, it currently belongs to the master branch. Of course, you can use v5.x.

Omnomios, 2024-05-20:
I used the following commands to get ESP-IDF 5.3
```
git pull --all
git checkout remotes/origin/release/v5.3
```
Then run the install.sh as normal.

Might be helpful to someone.