Xinyuan-LilyGO/LilyGo-Display-IDF #6: T-RGB 2.1 inch Half Circle: Trouble communicating with the screen

arthurABDesign, 2024-03-27:
Hello,
I can't seem to communicate with the screen after flashing the default code on my T-RGB.

While monitoring I get:

```log
[0;32mI (1098) main: ------ Initialize I2C.[0m
[0;33mW (1098) i2c.master: Please check pull-up resistances whether be connected properly. Otherwise unexpected behavior would happen. For more detailed information, please read docs[0m
[0;32mI (1098) gpio: GPIO[8]| InputEn: 1| OutputEn: 1| OpenDrain: 1| Pullup: 0| Pulldown: 0| Intr:0 [0m
[0;32mI (1098) gpio: GPIO[48]| InputEn: 1| OutputEn: 1| OpenDrain: 1| Pullup: 0| Pulldown: 0| Intr:0 [0m
[0;32mI (1098) main: ------ Initialize PMU.[0m
[0;32mI (1098) main: ------ Initialize TOUCH.[0m
[0;32mI (1098) main: ------ Initialize DISPLAY.[0m
[0;32mI (1098) RGB: ============T-RGB==============[0m
[0;31mE (1098) i2c.master: i2c_master_transmit_receive(1041): i2c handle not initialized[0m
[0;31mE (1098) i2c.master: i2c_master_transmit_receive(1041): i2c handle not initialized[0m
[0;31mE (1118) i2c.master: i2c_master_transmit_receive(1041): i2c handle not initialized[0m
[0;32mI (1128) RGB: Extension BEGIN SPI SUCCESS ![0m
[0;31mE (1128) i2c.master: i2c_master_transmit_receive(1041): i2c handle not initialized[0m
[0;31mE (1128) i2c.master: i2c_master_transmit_receive(1041): i2c handle not initialized[0m
[0;31mE (1128) i2c.master: i2c_master_transmit_receive(1041): i2c handle not initialized[0m
[0;31mE (1128) i2c.master: i2c_master_transmit_receive(1041): i2c handle not initialized[0m
[0;31mE (1128) i2c.master: i2c_master_transmit_receive(1041): i2c handle not initialized[0m
[0;31mE (1128) i2c.master: i2c_master_transmit_receive(1041): i2c handle not initialized[0m
[0;31mE (1128) i2c.master: i2c_master_transmit_receive(1041): i2c handle not initialized[0m
[0;31mE (1128) i2c.master: i2c_master_transmit_receive(1041): i2c handle not initialized[0m
[0;31mE (1128) i2c.master: i2c_master_transmit_receive(1041): i2c handle not initialized[0m
[0;31mE (1128) i2c.master: i2c_master_transmit_receive(1041): i2c handle not initialized[0m
[0;31mE (1128) i2c.master: i2c_master_transmit_receive(1041): i2c handle not initialized[0m
[0;31mE (1128) i2c.master: i2c_master_transmit_receive(1041): i2c handle not initialized[0m
[0;31mE (1128) i2c.master: i2c_master_transmit_receive(1041): i2c handle not initialized[0m
```

The `i2c.master: i2c_master_transmit_receive(1041): i2c handle not initialized`seems to come from the initialization of the screen in the `display_rgb.cpp`:

```c
  int i = 0;
  while (init_cmd[i].len != 0xff) {
      writeCommand(init_cmd[i].addr);
      writeData(init_cmd[i].param, init_cmd[i].len & 0x1F);
      if (init_cmd[i].len & 0x80) {
          delay(100);
      }
      i++;
  }
```

After a lot of repetitions of that error, the program seems to run, but the LCD stay black except for the backlight who light.

```
[0;31mE (2828) i2c.master: i2c_master_transmit_receive(1041): i2c handle not initialized[0m
[0;31mE (2828) i2c.master: i2c_master_transmit_receive(1041): i2c handle not initialized[0m
[0;31mE (2828) i2c.master: i2c_master_transmit_receive(1041): i2c handle not initialized[0m
[0;32mI (2948) RGB: Turn on LCD backlight[0m
[0;32mI (2948) gpio: GPIO[46]| InputEn: 0| OutputEn: 1| OpenDrain: 0| Pullup: 0| Pulldown: 0| Intr:0 [0m
[0;32mI (2948) main: Initialize LVGL library[0m
[0;32mI (2948) main: Register display driver to LVGL[0m
[0;32mI (2958) main: Install LVGL tick timer[0m
[0;32mI (2958) main: Register touch driver to LVGL[0m
[0;32mI (2958) main: Display LVGL[0m
[0;32mI (3248) main: Create LVGL task[0m
[0;32mI (3248) main_task: Returned from app_main()[0m
[0;32mI (3248) main: Starting LVGL task[0m
```

Did you managed to get the T-RGB configuration to work ?

I'm on the last version of the code [e458330](https://github.com/Xinyuan-LilyGO/LilyGo-Display-IDF/commit/e4583306f00cdd23e8d814745fcac36e0931d848) with the idf version `ESP-IDF v5.3-dev-2815-gbe06a6f5ff`.

Thanks

lewisxhe, 2024-07-25:
The I2C error has been fixed. This is the touch screen going into sleep mode, I2C access error, but the screen not lighting up situation does not exist. My side is working normally.

arthurABDesign, 2024-07-25:
Hello, the commit indeed fixed the issue. The screen is now working. Thanks

closed 2024-07-25 by arthurABDesign