Xinyuan-LilyGO/LilyGo-Display-IDF #10: Error building

SantiagoOlmosMed, 2024-07-24:
Hi, I'm kind of new to this, I'd like to know what I'm doing wrong, I'd appreciate your help!
`C:/Users/kidrl/OneDrive/Documentos/ESP-IDF-Proyects/NEW-AMOLED-IDF/IDF-AMOLED/main/display_rgb.cpp:15:10: fatal error: esp_lcd_panel_rgb.h: No such file or directory
   15 | #include "esp_lcd_panel_rgb.h"
      |          ^~~~~~~~~~~~~~~~~~~~~
compilation terminated.
[1409/1420] Building CXX object esp-idf/main/CMakeFiles/__idf_main.dir/power_driver.cpp.obj
In file included from C:/Users/kidrl/OneDrive/Documentos/ESP-IDF-Proyects/NEW-AMOLED-IDF/IDF-AMOLED/components/XPowersLib/src/PowersSY6970.tpp:35,
                 from C:/Users/kidrl/OneDrive/Documentos/ESP-IDF-Proyects/NEW-AMOLED-IDF/IDF-AMOLED/main/power_driver.cpp:92:
C:/Users/kidrl/OneDrive/Documentos/ESP-IDF-Proyects/NEW-AMOLED-IDF/IDF-AMOLED/components/XPowersLib/src/XPowersCommon.tpp: In instantiation of 'bool XPowersCommon<chipType>::begin(i2c_master_bus_handle_t, uint8_t) [with chipType = PowersSY6970; i2c_master_bus_handle_t = i2c_master_bus_t*; uint8_t = unsigned char]':
C:/Users/kidrl/OneDrive/Documentos/ESP-IDF-Proyects/NEW-AMOLED-IDF/IDF-AMOLED/main/power_driver.cpp:98:18:   required from here
C:/Users/kidrl/OneDrive/Documentos/ESP-IDF-Proyects/NEW-AMOLED-IDF/IDF-AMOLED/components/XPowersLib/src/XPowersCommon.tpp:171:29: warning: missing initializer for member 'i2c_device_config_t::scl_wait_us' [-Wmissing-field-initializers]
  171 |         i2c_device_config_t i2c_dev_conf = {
      |                             ^~~~~~~~~~~~
C:/Users/kidrl/OneDrive/Documentos/ESP-IDF-Proyects/NEW-AMOLED-IDF/IDF-AMOLED/components/XPowersLib/src/XPowersCommon.tpp:171:29: warning: missing initializer for member 'i2c_device_config_t::flags' [-Wmissing-field-initializers]
[1411/1420] Linking C static library esp-idf\lvgl__lvgl\liblvgl__lvgl.a
ninja: build stopped: subcommand failed.
Missing "esp_lcd_panel_rgb.h" file name found in the following component(s): esp_lcd(D:\esp-idf\v5.3\esp-idf\components\esp_lcd\rgb\include\esp_lcd_panel_rgb.h). Maybe one of the components needs to add the missing header directory to INCLUDE_DIRS of idf_component_register call in CMakeLists.txt.
ninja failed with exit code 1, output of the command is in the C:\Users\kidrl\OneDrive\Documentos\ESP-IDF-Proyects\NEW-AMOLED-IDF\IDF-AMOLED\build\log\idf_py_stderr_output_18920 and C:\Users\kidrl\OneDrive\Documentos\ESP-IDF-Proyects\NEW-AMOLED-IDF\IDF-AMOLED\build\log\idf_py_stdout_output_18920`

SantiagoOlmosMed, 2024-07-24:
Currently running on Windows 11, VS Code, ESP-IDF Extension v5.3
[image]

lewisxhe, 2024-07-25:
Everything compiles fine on ubuntu
[image]

SantiagoOlmosMed, 2024-07-25:
After a few unsuccessful attempts it compiled correctly, I appreciate your answer, thank you very much.
[image]

closed 2024-07-25 by SantiagoOlmosMed

n0xy1, 2025-01-07:
I know its a bit late, but the fix is to set the proper IDF_TARGET. 
Once set, esp-idf will add the proper headers. For this to work , my target was ESP32-S3